anurag3301/nvim-platformio.lua #6: Error loading with Packer

Erickrgb22, 2024-06-17:
Hi, i installed via Packer like the README says but when i execute any of the available commands, like Pioinit for example neovim gives me this error: 

[image]

Here is my packer.lua

[image]

Its any bad with my configs? or its really a bug or something weird?

anurag3301, 2024-08-23:
Hey @Erickrgb22 I forgot about this issue. Are you still facing this issue?

braedonwatkins, 2024-10-13:
I got this on my initial run but running again seemed to work. If it happens again I'll give more detailed info.

anurag3301, 2024-10-14:
Alr, closing the issue

closed 2024-10-14 by anurag3301