anurag3301/nvim-platformio.lua #9: Possibility of args completion

dhupee, 2024-11-09:
Thanks for making this plugin

one suggestion, is it possible for you to have args suggestions for the command, I'm surprised to see that the Piorun has args because

1. I don't really see the source code so I don't see that you can add
2. I use Wilder.nvim so usually it generate the possible arg but I don't see one so I don't know

if it's not possible or you don't have time but interested let me know, I can try to help

anurag3301, 2024-11-12:
That seems pretty cool idea, ill check it out

dhupee, 2024-11-12:
quick news, I've implement most of the command in lua, though I need to basically reimplement it in Lua to get the completion I want

https://github.com/dhupee/nvim-platformio.lua/blob/cmd-cmp/plugin/platformio.lua

but I'm still confuse how to implement Pioterm and Piodebug since I never used it, what kind of arguments it need anyway??

i also haven't saw it in your video in the README though I might missed it

anurag3301, 2024-11-13:
Hey, looks pretty good. 

> **TLDR:** you don't need auto suggestions for either of them

About Pioterm, its not a command. The actual command is `Piocmd` which you can use to run any arbitrary pio commands. `Piocmd` actually runs the file `pioterm.lua`. I have no idea why I kept a different name for this, will change it. You can check this
https://github.com/anurag3301/nvim-platformio.lua/blob/4b64bad524f22fa23f5a9480c4bd3e7a0c99e9ad/plugin/platformio.vim#L4

Now about how its used, this is and example 
https://github.com/anurag3301/nvim-platformio.lua/blob/4b64bad524f22fa23f5a9480c4bd3e7a0c99e9ad/doc/platformio.txt#L32-L42

So I guess you don't need any auto suggestions for this one as it's an arbitrary command.

About Piodebug, currently, it doesn't take any argument, its kind of in development, and I have not thought it thoroughly about how to setup debugging. if you look in the following code it does take an argument but it doesn't do anything with that. I kept it thinking maybe will do something later
https://github.com/anurag3301/nvim-platformio.lua/blob/4b64bad524f22fa23f5a9480c4bd3e7a0c99e9ad/plugin/platformio.vim#L7

dhupee, 2024-11-13:
so I use add PR on #10 if you are willing to check it before merge it to upstream, thanks

dhupee, 2024-11-13:
let's move to the PR

closed 2024-11-13 by dhupee