anurag3301/nvim-platformio.lua #17: Add Support For `clangd`

josephattalla, 2024-12-26:
Initializing creates a `.ccls` file, but I use `clangd`. To have my LSP work correctly I run `pio run -t compiledb` which creates a `compile_commands.json` file. It would be nice if there were an option to create this automatically when running `Pioinit`.

anurag3301, 2024-12-26:
Hey @josephattalla I just pushed a commit which addes a `setup` funtion which you can call from you neovim configuration. There you can set your lsp to `clangd` which is by defalult `ccls`. If you pick `clangd` it automatically   calles `pio run -t compiledb` and create the `compile_commands.json` file.

```lua
require('platformio').setup({
    lsp = "ccls" --default: ccls, other option: clangd
                 -- If you pick clangd, it also creates compile_commands.json
})
```

Let me know if it works for you

josephattalla, 2024-12-26:
Works, thanks

closed 2024-12-26 by josephattalla

anurag3301, 2025-04-01:
Hey @karshPrime  The plugin already generates `compile_commands.json` when you chose clangd as you lsp in config for this plugin.