anurag3301/nvim-platformio.lua #20: Can't activate language server

Tyrn, 2025-04-11:
Hi,

I've been trying to create a ESP32 project on Arduino framework. My fault, probably, but I can't figure it out myself.

My project, created via `:Pioinit`:

`esp_project/platformio.ini`:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```
Platform installed via `pio pkg install -p "platformio/espressif32"`; `locate Arduino.h`:
```
/home/alexey/.arduino15/packages/STMicroelectronics/hardware/stm32/2.10.1/cores/arduino/Arduino.h
/home/alexey/.arduino15/packages/esp32/hardware/esp32/3.2.0/cores/esp32/Arduino.h
/home/alexey/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h
```
The language server didn't even see `Arduino.h` file. I checked issue [#17](https://github.com/anurag3301/nvim-platformio.lua/issues/17), and tried
```
pio run -t compiledb
```
Now `compile_commands.json` is present and `Arduino.h` is visible, but it doesn't look like the right one. I've got a lot of errors:
```
clang: Unknown argument '-mlongcalls'; did you mean '-mlong-calls'? [drv_unknown_argument_with_suggestion]
clang: Unknown argument: '-fno-tree-switch-conversion' [drv_unknown_argument]
clang: Unknown argument: '-fstrict-volatile-bitfields' [drv_unknown_argument]
clang: In included file: 'machine/endian.h' file not found [pp_file_not_found]
```
`~/.config/nvim/lua/plugins/nvim-platformio.lua`:
```
return {
	-- PlatformIO.
	{
		"anurag3301/nvim-platformio.lua",
		dependencies = {
			{ "akinsho/nvim-toggleterm.lua" },
			{ "nvim-telescope/telescope.nvim" },
			{ "nvim-lua/plenary.nvim" },
		},
		lazy = false,
		init = function()
			require("platformio").setup({
				lsp = "clangd", --default: ccls, other option: clangd
				-- If you pick clangd, it also creates compile_commands.json
			})
		end,
		-- event = "BufRead",
	},
}
```
Actually, I use AstroNvim. I suspect the `require` clause isn't working here, it's just the community cpp plugin which is picking something else. Unfortunately, I don't know where to start my research...

Status line shows `clangd`, regardless of the choice inside `plugins/nvim-platformio.lua`.

anurag3301, 2025-04-13:
Hey @Tyrn, I am not sure about AstroNvim, will test it on later. For now I tried on a minimal_config with clangd it it does works.

```lua
-- insures lazy is installed
local lazypath = vim.loop.os_tmpdir() .. "/lazy/lazy.nvim"
if not (vim.uv or vim.loop).fs_stat(lazypath) then
  vim.fn.system({
    "git",
    "clone",
    "--filter=blob:none",
    "https://github.com/folke/lazy.nvim.git",
    "--branch=stable", -- latest stable release
    lazypath,
  })
end
vim.opt.rtp:prepend(lazypath)

local plugins = {
    {
        "anurag3301/nvim-platformio.lua",
        dependencies = {
            { "akinsho/nvim-toggleterm.lua" },
            { "nvim-telescope/telescope.nvim" },
            { "nvim-lua/plenary.nvim" },
        },
    },
    {
        "neovim/nvim-lspconfig",
        'hrsh7th/cmp-nvim-lsp',
        'hrsh7th/cmp-buffer',
        'hrsh7th/nvim-cmp',
    },
}

require("lazy").setup(plugins, {
    install = {
        missing = true,
    },
})

vim.opt['number'] = true
require('platformio').setup({
    lsp = "clangd" --default: ccls, other option: clangd
                 -- If you pick clangd, it also creates compile_commands.json
})

local nvim_lsp = require('lspconfig')

local capabilities = vim.lsp.protocol.make_client_capabilities()
capabilities = require('cmp_nvim_lsp').default_capabilities(capabilities)

nvim_lsp["clangd"].setup {
    on_attach = on_attach,
    capabilities = capabilities,
    flags = {
        debounce_text_changes = 150,
    },
    root_dir = function() return vim.loop.cwd() end
}

local cmp = require'cmp'

cmp.setup({
sources = cmp.config.sources({
  { name = 'nvim_lsp' },
}, {
  { name = 'buffer' },
})
})

```

anurag3301, 2025-04-13:
Repeat following steps
1. Copy this code and save it as `minimal_config.lua`.
2. Open neovim with this command `nvim -u minimal_config.lua`
3. Run `:PioInit` and select your board, In my case I picked `Espressif ESP32 Dev Module`
4. Pick Arduino as framework, then it will create a project
5. Open a new main.cpp file by running `:e src/main.cpp`
6. Now run `:LspInfo` and it should show following infomation about lsp 

[image]

7. Come back to main.cpp now if you start typing it should detech Arduino.h 

[image]

> Let me know if you are able to reproduce it, then ill check whats going on with AstroNvim

Tyrn, 2025-04-13:
Thanks!

Yes, I can reproduce it without a hitch.

anurag3301, 2025-04-14:
If you can reproduce it then it works, prob something wrong with astronvim config

Tyrn, 2025-04-14:
>  If you can reproduce it then it works, prob something wrong with astronvim config

Of course it is. The whole point of using AstroNvim is the possibility of having a lot of common plugins preconfigured and ready for use. All it takes to configure C++ is two lines in `nvim/lua/community.lua`:
```
---@type LazySpec
return {
	"AstroNvim/astrocommunity",
	{ import = "astrocommunity.recipes.neovide" },
	{ import = "astrocommunity.pack.lua" },
	{ import = "astrocommunity.pack.cmake" },
	{ import = "astrocommunity.pack.cpp" },
...
```
You'll have to pay for this convenience when some customization is in order. There's also a file dedicated to lsp configuration, `nvim/lua/plugins/astrolsp.lua` (note the first line; it isn't even enabled):
```
if true then return {} end -- WARN: REMOVE THIS LINE TO ACTIVATE THIS FILE

-- AstroLSP allows you to customize the features in AstroNvim's LSP configuration engine
-- Configuration documentation can be found with `:h astrolsp`
-- NOTE: We highly recommend setting up the Lua Language Server (`:LspInstall lua_ls`)
--       as this provides autocomplete and documentation while editing

---@type LazySpec
return {
  "AstroNvim/astrolsp",
  ---@type AstroLSPOpts
  opts = {
    -- Configuration table of features provided by AstroLSP
    features = {
      codelens = true, -- enable/disable codelens refresh on start
      inlay_hints = false, -- enable/disable inlay hints on start
      semantic_tokens = true, -- enable/disable semantic token highlighting
    },
    -- customize lsp formatting options
    formatting = {
      -- control auto formatting on save
      format_on_save = {
        enabled = true, -- enable or disable format on save globally
        allow_filetypes = { -- enable format on save for specified filetypes only
          -- "go",
        },
        ignore_filetypes = { -- disable format on save for specified filetypes
          -- "python",
        },
      },
      disabled = { -- disable formatting capabilities for the listed language servers
        -- disable lua_ls formatting capability if you want to use StyLua to format your lua code
        -- "lua_ls",
      },
      timeout_ms = 1000, -- default format timeout
      -- filter = function(client) -- fully override the default formatting function
      --   return true
      -- end
    },
    -- enable servers that you already have installed without mason
    servers = {
      -- "pyright"
    },
    -- customize language server configuration options passed to `lspconfig`
    ---@diagnostic disable: missing-fields
    config = {
      -- clangd = { capabilities = { offsetEncoding = "utf-8" } },
    },
    -- customize how language servers are attached
    handlers = {
      -- a function without a key is simply the default handler, functions take two parameters, the server name and the configured options table for that server
      -- function(server, opts) require("lspconfig")[server].setup(opts) end

      -- the key is the server that is being setup with `lspconfig`
      -- rust_analyzer = false, -- setting a handler to false will disable the set up of that language server
      -- pyright = function(_, opts) require("lspconfig").pyright.setup(opts) end -- or a custom handler function can be passed
    },
    -- Configure buffer local auto commands to add when attaching a language server
    autocmds = {
      -- first key is the `augroup` to add the auto commands to (:h augroup)
      lsp_codelens_refresh = {
        -- Optional condition to create/delete auto command group
        -- can either be a string of a client capability or a function of `fun(client, bufnr): boolean`
        -- condition will be resolved for each client on each execution and if it ever fails for all clients,
        -- the auto commands will be deleted for that buffer
        cond = "textDocument/codeLens",
        -- cond = function(client, bufnr) return client.name == "lua_ls" end,
        -- list of auto commands to set
        {
          -- events to trigger
          event = { "InsertLeave", "BufEnter" },
          -- the rest of the autocmd options (:h nvim_create_autocmd)
          desc = "Refresh codelens (buffer)",
          callback = function(args)
            if require("astrolsp").config.features.codelens then vim.lsp.codelens.refresh { bufnr = args.buf } end
          end,
        },
      },
    },
    -- mappings to be set up on attaching of a language server
    mappings = {
      n = {
        -- a `cond` key can provided as the string of a server capability to be required to attach, or a function with `client` and `bufnr` parameters from the `on_attach` that returns a boolean
        gD = {
          function() vim.lsp.buf.declaration() end,
          desc = "Declaration of current symbol",
          cond = "textDocument/declaration",
        },
        ["<Leader>uY"] = {
          function() require("astrolsp.toggles").buffer_semantic_tokens() end,
          desc = "Toggle LSP semantic highlight (buffer)",
          cond = function(client)
            return client.supports_method "textDocument/semanticTokens/full" and vim.lsp.semantic_tokens ~= nil
          end,
        },
      },
    },
    -- A custom `on_attach` function to be run after the default `on_attach` function
    -- takes two parameters `client` and `bufnr`  (`:h lspconfig-setup`)
    on_attach = function(client, bufnr)
      -- this would disable semanticTokensProvider for all clients
      -- client.server_capabilities.semanticTokensProvider = nil
    end,
  },
}
```
I tried to comment out the first line, and uncomment the
```
    config = {
      -- clangd = { capabilities = { offsetEncoding = "utf-8" } },
    },
```
but nothing changed. It isn't that easy :)

Here's the so called cpp community [pack](https://github.com/AstroNvim/astrocommunity/blob/main/lua/astrocommunity/pack/cpp/init.lua). That's probably all there's to it. Unfortunately, I don't know what to try.

Tyrn, 2025-04-14:
Yes, it's active:
```

==============================================================================
vim.lsp:                                     require("vim.lsp.health").check()

- LSP log level : WARN
- Log path: /home/alexey/.local/state/nvim/lsp.log
- Log size: 1912 KB

vim.lsp: Active Clients ~
- clangd (id: 1)
  - Version: clangd version 20.1.0 (https://github.com/llvm/llvm-project 24a30daaa559829ad079f2ff7f73eb4e18095f88) linux+grpc x86_64-unknown-linux-gnu
  - Root directory: ~/spaces/cpp/anurag
  - Command: { "/home/alexey/.local/share/nvim/mason/bin/clangd" }
  - Settings: vim.empty_dict()
  - Attached buffers: 4

vim.lsp: Enabled Configurations ~

vim.lsp: File Watcher ~
- file watching "(workspace/didChangeWatchedFiles)" disabled on all clients

vim.lsp: Position Encodings ~
- No buffers contain mixed position encodings

```
It stands to reason: one has to show me the errors which shouldn't be there.

anurag3301, 2025-04-15:
Does your LSP works in general, If you open a random cpp file and start typing, does it give you autocompletions? If yes then does it give any sort of autocompletion in platformio project.

Tyrn, 2025-04-15:
Plain C:
```
#include <stdio.h>

int main() { printf("%s\n", "Hello, World"); }
```
Manual input, with every conceivable prompt and warning.

PlatformIO:

Yes, it gives autocompletion to `#include <Arduino.h>`. Then a flurry of errors:
```
clang: Unknown argument '-mlongcalls'; did you mean '-mlong-calls'? [drv_unknown_argument_with_suggestion]
clang: Unknown argument: '-fno-tree-switch-conversion' [drv_unknown_argument]
clang: Unknown argument: '-fstrict-volatile-bitfields' [drv_unknown_argument]
clang: In included file: 'machine/endian.h' file not found [pp_file_not_found]
```
As I understand it,  the language server runs, but on the wrong library or something.

anurag3301, 2025-04-18:
It seems there is problem with the `compile_commands.json`  generated by platformio and these are diagnostic warnings. If you dont want them you can configure your diagnostic warnings. Rest should be fine, it should be giving all the arduino function autocompletions

Tyrn, 2025-04-18:
`File not found` doesn't look like a warning. Besides, on
```
  Serial.println("Hello world!");
```
it's a fatal error
```
clang: No matching member function for call to 'println' [ovl_no_viable_member_function_in_call]
```

As to `compile_commands.json`: generated by your plugin "navitely" via `nvim -u minimal_config.lua` and generated with `pio run -t compiledb` are indeed slightly different, but the behavior is exactly the same.

anurag3301, 2025-04-19:
Nope, my plugin does not produce any `compile_commands.json` natively, its `pio core` who produce it. 

I suppose you are talking about these following errors.

[image]

These are still not compile error, these are lsp diagnostics which is caused due to configurations in `compile_commands.json` created by the `pio core` for `esp 32`. Its the same problem with any esp board, be it 8266.

[image]

It does not produce such LSP diagnostic error for other platform such as AVR, STM32 etc. I am not sure what can I do on my plugin end to fix these errors because its a problem between `pio core` and `clangd`. You can just turn off LSP diagnostics  and you wont see those errors popping up in your editor. I use [WhoIsSethDaniel/toggle-lsp-diagnostics.nvim](https://github.com/WhoIsSethDaniel/toggle-lsp-diagnostics.nvim)

Tyrn, 2025-04-19:
Thank you very much! Fantastic. It never occurred to me that one can live with messages like `File not found` or `No matching member function`. But actually you can, and even have autocomplete for those nonexistent functions :)

No `toggle-lsp-diagnostics.nvim` is needed, just as they promised. With AstroNvim, there's `<Leader>ud` specifically for the purpose.

anurag3301, 2025-04-20:
Alright, I've got a fix for your problem. Again the problem lies under the `compile_command.json` generated by pio code and how its interpreted by `clangd`. A C/C++ LSP ideally only need 2 things, the `-D` define flags and `-I` fiags for the include paths.  With the help of ChatGPT I came up with a python  script which does follwing.
1. Runs `pio run -t compiledb` to generate original `compile_commands.json` from pio code
2. Renames it so it can be used later to extract path to the actual compiler path
3. Then it uses the `.ccls` file to extract **only the -D and -I flags** and create a new very clean `compile_commands.json` 

And I think it works perfect, no more lsp errors or warning, You dont have to turn of lsp diagnostics anymore and the autocompletions works as it should. You can look at follwing screenshot. No more errors related to `Arduino.h` and `Serial.println`

[image]

Now if you want to try it yourself, just create a python file in root directory of your project called `convert.py` and paste following code in it

```py
import json
from pathlib import Path
import subprocess

# Constants
PROJECT_ROOT = Path.cwd()
SRC_FILE = PROJECT_ROOT / "src" / "main.cpp"
CCDB_ORIG = PROJECT_ROOT / "compile_commands.json"
CCDB_BAK = PROJECT_ROOT / "compile_commands.json.bak"
CCLS_FILE = PROJECT_ROOT / ".ccls"
CCDB_NEW = PROJECT_ROOT / "compile_commands.json"

KEEP_FLAGS = ("-I", "-D", "-std", "-Wall")

def ensure_compile_commands():
    if not CCDB_ORIG.exists():
        subprocess.run(["pio", "run", "-t", "compiledb"], check=True)
    if not CCDB_BAK.exists():
        CCDB_ORIG.rename(CCDB_BAK)

def extract_compiler():
    ccdb_data = json.loads(CCDB_BAK.read_text())
    if not ccdb_data:
        raise RuntimeError("Empty compile_commands.json.bak")
    cmd = ccdb_data[0]["command"]
    compiler = cmd.split()[0]
    return compiler

def parse_ccls():
    common_flags = []
    cpp_flags = []
    with open(CCLS_FILE) as f:
        for line in f:
            line = line.strip()
            if not line or line.startswith("#"):
                continue
            if line.startswith("%cpp"):
                cpp_flags.extend([flag for flag in line.split()[1:] if flag == "-Wall" or flag.startswith(KEEP_FLAGS)])
            elif line.startswith("%c"):
                pass  # skip C files for now
            elif not line.startswith("%"):
                common_flags.extend([flag for flag in line.split() if flag == "-Wall" or flag.startswith(KEEP_FLAGS)])
    return common_flags + cpp_flags

def generate_single_entry_compile_command(compiler, flags):
    entry = {
        "directory": str(PROJECT_ROOT),
        "file": str(SRC_FILE),
        "command": f"{compiler} {' '.join(flags)} {SRC_FILE}"
    }
    CCDB_NEW.write_text(json.dumps([entry], indent=2))
    return 1

# Execution
ensure_compile_commands()
compiler_path = extract_compiler()
filtered_flags = parse_ccls()
entry_count = generate_single_entry_compile_command(compiler_path, filtered_flags)

entry_count
```

And just run `python convert.py` and then reopen neovim and it should be fine now. Let me know it works.

#### If it sounds like a valid fix I could integrated it in the plugin itself. What do you think @batoaqaa @dhupee

Tyrn, 2025-04-20:
Excellent! Yes, it works.

anurag3301, 2025-04-20:
Great closing the issue!

closed 2025-04-20 by anurag3301

Tyrn, 2025-04-20:
Are you planning to incorporate the know-how into the plugin?

dhupee, 2025-04-21:
> Alright, I've got a fix for your problem. Again the problem lies under the `compile_command.json` generated by pio core and how its interpreted by `clangd`. A C/C++ LSP ideally only need 2 things, the `-D` define flags and `-I` fiags for the include paths. With the help of ChatGPT I came up with a python script which does follwing.
> 
>     1. Runs `pio run -t compiledb` to generate original `compile_commands.json` from pio core
> 
>     2. Renames it so it can be used later to extract path to the actual compiler path
> 
>     3. Then it uses the `.ccls` file to extract **only the -D and -I flags** and create a new very clean `compile_commands.json`
> 
> 
> And I think it works perfect, no more lsp errors or warning, You dont have to turn of lsp diagnostics anymore and the autocompletions works as it should. You can look at follwing screenshot. No more errors related to `Arduino.h` and `Serial.println`
> 
> [image]
> 
> Now if you want to try it yourself, just create a python file in root directory of your project called `convert.py` and paste following code in it
> 
> import json
> from pathlib import Path
> import subprocess
> 
> # Constants
> PROJECT_ROOT = Path.cwd()
> SRC_FILE = PROJECT_ROOT / "src" / "main.cpp"
> CCDB_ORIG = PROJECT_ROOT / "compile_commands.json"
> CCDB_BAK = PROJECT_ROOT / "compile_commands.json.bak"
> CCLS_FILE = PROJECT_ROOT / ".ccls"
> CCDB_NEW = PROJECT_ROOT / "compile_commands.json"
> 
> KEEP_FLAGS = ("-I", "-D", "-std", "-Wall")
> 
> def ensure_compile_commands():
>     if not CCDB_ORIG.exists():
>         subprocess.run(["pio", "run", "-t", "compiledb"], check=True)
>     if not CCDB_BAK.exists():
>         CCDB_ORIG.rename(CCDB_BAK)
> 
> def extract_compiler():
>     ccdb_data = json.loads(CCDB_BAK.read_text())
>     if not ccdb_data:
>         raise RuntimeError("Empty compile_commands.json.bak")
>     cmd = ccdb_data[0]["command"]
>     compiler = cmd.split()[0]
>     return compiler
> 
> def parse_ccls():
>     common_flags = []
>     cpp_flags = []
>     with open(CCLS_FILE) as f:
>         for line in f:
>             line = line.strip()
>             if not line or line.startswith("#"):
>                 continue
>             if line.startswith("%cpp"):
>                 cpp_flags.extend([flag for flag in line.split()[1:] if flag == "-Wall" or flag.startswith(KEEP_FLAGS)])
>             elif line.startswith("%c"):
>                 pass  # skip C files for now
>             elif not line.startswith("%"):
>                 common_flags.extend([flag for flag in line.split() if flag == "-Wall" or flag.startswith(KEEP_FLAGS)])
>     return common_flags + cpp_flags
> 
> def generate_single_entry_compile_command(compiler, flags):
>     entry = {
>         "directory": str(PROJECT_ROOT),
>         "file": str(SRC_FILE),
>         "command": f"{compiler} {' '.join(flags)} {SRC_FILE}"
>     }
>     CCDB_NEW.write_text(json.dumps([entry], indent=2))
>     return 1
> 
> # Execution
> ensure_compile_commands()
> compiler_path = extract_compiler()
> filtered_flags = parse_ccls()
> entry_count = generate_single_entry_compile_command(compiler_path, filtered_flags)
> 
> entry_count
> 
> And just run `python convert.py` and then reopen neovim and it should be fine now. Let me know it works.
> #### If it sounds like a valid fix I could integrated it in the plugin itself. What do you think [@batoaqaa](https://github.com/batoaqaa) [@dhupee](https://github.com/dhupee)

I would be honest to you that I always deletes the `.ccls` file then I use this bash script from https://skylab.hashnode.dev/platformio-with-neovim that generates the files I need and  then I'm able to get LSP which is `clangd` up and running, if I use `.ccls` I only able to get basic C++ LSP but not Pio library specific LSP

This is my [dotfiles](https://github.com/dhupee/dotfiles/tree/master/private_dot_nix-configs/private_config/private_nvim) if you want to see

batoaqaa, 2025-04-21:
per Clangd Getting started (https://clangd.llvm.org/installation) and
Compilation database compile_commands.json (
https://docs.platformio.org/en/latest/integration/compile_commands.html)
I believe you can have better result of the convert.py script by renaming
".ccls" file to "compile_flags.txt" and rename "compile_commands.json" to "
compile_commands.json.back" as  "compile_flags.txt" will be ignored if
compile_commands.json is present.
This is OK If all files in a project use the same build flags. However
background-indexing will not work, as clangd can’t be sure which of the
files are project sources.

Using compile_command.json  (
https://docs.platformio.org/en/latest/integration/compile_commands.html)
created with:

platformio.ini:

[env:myenv]platform = ...board = ...extra_scripts = pre:extra_script.py
_____________

extra_script.py:

import osImport("env")
# include toolchain pathsenv.Replace(COMPILATIONDB_INCLUDE_TOOLCHAIN=True)
# override compilation DB
pathenv.Replace(COMPILATIONDB_PATH=os.path.join("$BUILD_DIR",
"compile_commands.json"))
___________

Generate compile_commands.json

> pio run -t compiledb

This is a lot better as it will recursively traverse your project
files, all platform files and libraries files.

Regards,

anurag3301, 2025-04-21:
@batoaqaa your solution is perfect, didnt know you could just use `.ccls` file for clangd. I am thinking to integrated it in the plugin to rename `.ccls` to `compile_flags.txt` if the user has configured `lsp = "ccls"`. Now the question is, is there any need for `compile_commands.json`

dhupee, 2025-04-21:
> [@batoaqaa](https://github.com/batoaqaa) your solution is perfect, didnt know you could just use `.ccls` file for clangd. I am thinking to integrated it in the plugin to rename `.ccls` to [...]

I don't think its a bad idea just to leave it, at least until can confirm whether both `ccls` and `clangd` wont have issue if its missing

anurag3301, 2025-04-21:
> I don't think its a bad idea just to leave it, at least until can confirm whether both `ccls` and `clangd` wont have issue if its missing

The problem is if I leave `compile_commands.json` as it is and also produce `compile_flags.txt` for `clangd`, `clangd` which prefer `compile_commands.json` instead of `compile_flags.txt` leaving the user with same problem.

dhupee, 2025-04-21:
> The problem is if I leave `compile_commands.json` as it is and also produce `compile_flags.txt` for `clangd`, `clangd` which prefer `compile_commands.json` instead of `compile_flags.txt` leaving the [...]

Let's try your way but in seperate branch, not disturbing 'master' branch, verifying that your method works, until then you merge it, it's your project anyway

anurag3301, 2025-04-21:
Sounds good

Tyrn, 2025-04-21:
Since it's all about [build flags](https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html)... Does it make sense to use this option?

Maybe the idea to excise the unnecessary ones is better, I can't judge by my limited experience.