anurag3301/nvim-platformio.lua #23: Bug

Naster17, 2025-06-01:
[image]
This only applies to Linter as the assembly and installation is successful

```
clangd
platform = espressif32
board = esp32dev
framework = arduino
```

[compile_commands.json](https://github.com/user-attachments/files/20542540/compile_commands.json)

Naster17, 2025-06-01:
[image]
also if create new project

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

anurag3301, 2025-06-03:
This is duplicate of #20 , I recommend you read the thread for more information. 

TLDR: run following in the project root directory where you have `platformio.ini` file 

```sh
rm compile_commands.json
curl -s https://anurag3301.com/files/GrDSa.py | python3
```

This will run a python script which you can find here https://github.com/anurag3301/nvim-platformio.lua/issues/20#issuecomment-2817088342 to create a new sanitized `compile_commands.json`

Let me know if you face any problem

closed 2025-06-03 by Naster17

Tyrn, 2025-06-09:
I'm sorry. The script requires `.ccls` file, but somehow I can't generate it with my new project. I just can't remember how I got first time two months ago :(

anurag3301, 2025-06-11:
Yes the script requires `.ccls` file from which the python script extracts the `-D`, `-I` and other necessary flags and create a new `compile_commands.json` file which clangd can use without any problem. The ccls file can be easily be generate using `--ide=vim` flag.

Creating `compile_command.json` file from `ccls` does feel bit hacky but not sure any other way to deal with this problem.