anurag3301/nvim-platformio.lua #25: STM32 and clangd

Tyrn, 2025-06-19:
I tried to build a simple project with STM32. I employed your `fix_commandcompile.py` file which is good specifically for ESP32, as I understand. With STM32 the result is less than perfect:

[image]

What should I do make it correct?

My essential files:

`platformio.ini`:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = hid
build_flags =
  -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
```
`configure.sh`:
```
#/usr/bin/env bash

# Configure/reconfigure the project and the language server.

pio project init --ide=vim
pio run -t compiledb
#curl -s https://anurag3301.com/files/GrDSa.py | python
python fix_cmdcompile.py
```
The project gets built and uploaded fine.

anurag3301, 2025-06-20:
I dont think you need to run the script for stm32, the default works totally fine. The script was made for ESPs.

[image]

And about the last line, it was just for REPL, doesn't do anything

anurag3301, 2025-06-23:
Closing it then

closed 2025-06-23 by anurag3301