anurag3301/nvim-platformio.lua #39: Formatting discrepancy

batoaqaa, 2025-08-11:
I noticed that there is discrepancy between some files and '.stylua.toml' formatting.

For example file '\nvim-platformio.lua\lua\platformio\utils.lua' was using quote style as double, but '.stylua.toml' using quote_style as "AutoPreferSingle". This way any time you make small changes to the file and save the file, formatting will go and do a lot of changes just to convert quote from single to double.

I believe this because those files were saved before you use '.stylua.toml' formatting. 

I suggest you go through all the files and edit then save them to unify formatting across all the files.

anurag3301, 2025-08-11:
Yes, sure ill do that, I think `.stylua.toml` doesnt even take effect on my nvim, it picks up defaults from my nvim config like 4 space tabs. Do I need to install some plugin to let it pickup the `.stylua.toml` file

closed 2025-08-11 by anurag3301

anurag3301, 2025-08-12:
Great thanks, I'll set it up